ros2/rmw_cyclonedds #330: Windows 10 - build issue with unresolved external symbol in serdata.hpp

jonathanTIE, 2021-08-06:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10, build 17763
- Installation type:
  - binaries (but trying to build rmw_cyclonedds)
- Version or commit hash:
  - branch galactic, but same problem with master
- DDS implementation:
  - cycloneDDS

#### Steps to reproduce issue
I'm trying to build nav2 for windows 10, and I need to build rclcpp, which in turns need rmw_cyclonedds.
All dependencies below build fine (after some troubleshooting), the only package failing and blocking the others is this one.

```
colcon build --packages-up-to rmw_cyclonedds_cpp --base-paths src --event-handlers console_direct+
```

#### Expected behavior

Build should work

#### Actual behavior

```

Starting >>> rmw_cyclonedds_cpp
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.17763.
-- Found ament_cmake_ros: 0.9.2 (C:/opt/ros/galactic/x64/install/ament_cmake_ros/share/ament_cmake_ros/cmake)
-- Using PYTHON_EXECUTABLE: C:/opt/ros/galactic/x64/python.exe
-- Found rcutils: 4.0.2 (C:/opt/ros/galactic/x64/install/rcutils/share/rcutils/cmake)
-- Found rcpputils: 2.2.0 (C:/opt/ros/galactic/x64/install/rcpputils/share/rcpputils/cmake)
-- Found rmw: 3.3.1 (C:/opt/ros/galactic/x64/install/rmw/share/rmw/cmake)
-- Found rmw_dds_common: 1.2.1 (C:/opt/ros/galactic/x64/install/rmw_dds_common/share/rmw_dds_common/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Found rosidl_adapter: 2.2.1 (C:/opt/ros/galactic/x64/install/rosidl_adapter/share/rosidl_adapter/cmake)
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_introspection_cpp;rosidl_typesupport_fastrtps_cpp
-- Configuring done
-- Generating done
-- Build files have been written to: C:/opt/ros/galactic/x64/build/rmw_cyclonedds_cpp
Microsoft (R) Build Engine version 16.6.0+5ff7b0c9e for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Serialization.cpp
  rmw_node.cpp
  serdata.cpp
     Creating library C:/opt/ros/galactic/x64/build/rmw_cyclonedds_cpp/Release/rmw_cyclonedds_cpp.lib and object C:/opt/ros/galactic/x64/build/rmw_cyclonedds_cpp/Release/rmw_cyclonedds_cpp.exp
rmw_node.obj : error LNK2019: unresolved external symbol __imp_dds_create_topic_sertype referenced in function "struct CddsPublisher * __cdecl create_cdds_publisher(int,int,struct rosidl_message_type_support_t const *,char const *,struct rmw_qos_profile_t const *)" (?create_cdds_publisher@@YAPEAUCddsPublisher@@HHPEBUrosidl_message_type_support_t@@PEBDPEBUrmw_qos_profile_t@@@Z) [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]
rmw_node.obj : error LNK2019: unresolved external symbol __imp_ddsi_sertype_unref referenced in function "struct CddsPublisher * __cdecl create_cdds_publisher(int,int,struct rosidl_message_type_support_t const *,char const *,struct rmw_qos_profile_t const *)" (?create_cdds_publisher@@YAPEAUCddsPublisher@@HHPEBUrosidl_message_type_support_t@@PEBDPEBUrmw_qos_profile_t@@@Z) [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]
serdata.obj : error LNK2019: unresolved external symbol __imp_ddsi_sertype_init referenced in function "struct sertype_rmw * __cdecl create_sertype(char const *,char const *,void *,bool,class std::unique_ptr<class rmw_cyclonedds_cpp::StructValueType,struct std::default_delete<class rmw_cyclonedds_cpp::StructValueType> >,unsigned int,bool)" (?create_sertype@@YAPEAUsertype_rmw@@PEBD0PEAX_NV?$unique_ptr@VStructValueType@rmw_cyclonedds_cpp@@U?$default_delete@VStructValueType@rmw_cyclonedds_cpp@@@std@@@std@@I_N@Z) [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]
serdata.obj : error LNK2019: unresolved external symbol __imp_ddsi_sertype_fini referenced in function "void __cdecl sertype_rmw_free(struct ddsi_sertype *)" (?sertype_rmw_free@@YAXPEAUddsi_sertype@@@Z) [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]
C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\Release\rmw_cyclonedds_cpp.dll : fatal error LNK1120: 4 unresolved externals [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]
Failed   <<< rmw_cyclonedds_cpp [8.23s, exited with code 1]

```

#### Additional information

I tried to narrow down the issue, in `serdata.hpp` , it seems to be linked with these (lines 30-34):
```

#if !DDS_HAS_DDSI_SERTYPE
#define ddsi_sertype ddsi_sertopic
#define ddsi_sertype_ops ddsi_sertopic_ops
#define sertype_rmw sertopic_rmw
#define sertype_rmw_ops sertopic_rmw_ops
#endif

```

Commenting the if and endif just lead to more error :

```
Microsoft (R) Build Engine version 16.6.0+5ff7b0c9e for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Serialization.cpp
  rmw_node.cpp
  serdata.cpp
C:\opt\ros\galactic\x64\src\ros2\rmw_cyclonedds\rmw_cyclonedds_cpp\src\serdata.hpp(48,1): error C2504: 'ddsi_sertopic': base class undefined (compiling source file C:\opt\ros\galactic\x64\src\ros2\rmw_cyclonedds\rmw_cyclonedds_cpp\src\serdata.cpp) [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]
C:\opt\ros\galactic\x64\src\ros2\rmw_cyclonedds\rmw_cyclonedds_cpp\src\serdata.hpp(48,1): error C2504: 'ddsi_sertopic': base class undefined (compiling source file C:\opt\ros\galactic\x64\src\ros2\rmw_cyclonedds\rmw_cyclonedds_cpp\src\Serialization.cpp) [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]
C:\opt\ros\galactic\x64\src\ros2\rmw_cyclonedds\rmw_cyclonedds_cpp\src\serdata.cpp(117,84): error C2440: 'static_cast': cannot convert from 'const ddsi_sertype *' to 'const sertopic_rmw *' [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]
C:\opt\ros\galactic\x64\src\ros2\rmw_cyclonedds\rmw_cyclonedds_cpp\src\serdata.cpp(117,37): message : Types pointed to are unrelated; conversion requires reinterpret_cast, C-style cast or function-style cast [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]
C:\opt\ros\galactic\x64\src\ros2\rmw_cyclonedds\rmw_cyclonedds_cpp\src\serdata.hpp(48,1): error C2504: 'ddsi_sertopic': base class undefined (compiling source file C:\opt\ros\galactic\x64\src\ros2\rmw_cyclonedds\rmw_cyclonedds_cpp\src\rmw_node.cpp) [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]
C:\opt\ros\galactic\x64\src\ros2\rmw_cyclonedds\rmw_cyclonedds_cpp\src\serdata.cpp(232,86): error C2440: 'static_cast': cannot convert from 'const ddsi_sertopic *' to 'const sertopic_rmw *' [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]
C:\opt\ros\galactic\x64\src\ros2\rmw_cyclonedds\rmw_cyclonedds_cpp\src\serdata.cpp(232,39): message : Types pointed to are unrelated; conversion requires reinterpret_cast, C-style cast or function-style cast [C:\opt\ros\galactic\x64\build\rmw_cyclonedds_cpp\rmw_cyclonedds_cpp.vcxproj]

[......]

```

eboasson, 2021-08-07:
I agree, builds should work. And, looking at the ROS 2 build farm, they generally do!

If I were to guess: the compiler is using the header files from the correct/current version of Cyclone DDS, but the linker is trying to link against an old version. That's why at first you only had the linker complain about missing symbols, while with your `DDS_HAS_DDSI_SERTYPE` experiment, you got error messages from the compiler about type definitions.

The introduction of `ddsi_sertype` in Cyclone DDS dates to November 2020, which makes it reasonably ancient by now. Cyclone DDS releases tend to be (too) rare, so perhaps you're being bitten by having a Cyclone DDS 0.7.0 binary or so, possibly installed as part of, e.g., Foxy, in the path. 

On Linux, I have had some difficulty myself with having a ROS 2 release installed from binaries and then trying to do a `colcon` build of just the Cyclone DDS parts in an overlay workspace, because the paths set in the environment would cause the build to pick up some components (and only some!) from the binary packages rather than from the freshly built ones. Not exactly the same, I know, but the errors you describe are definitely similar.

In short, the best place to start looking is your PATH and linker path settings and whether you did a `setup.bat` for, e.g., Foxy, in the command line window prior to starting the build.

jonathanTIE, 2021-08-07:
Thanks for the detailed answer ! 
I think you are right, I actually used binaries from ms IoT to install ros, and the version I used had many outdated dependencies. It may have tried to use these one instead of the newly built one. I resintalled ros with working dependencies and nav2 seems to work, so I don't need to build cyclonedds, thanks for your time !

closed 2021-08-07 by jonathanTIE