ros2/rmw_cyclonedds #331: Missing quality declaration for rmw_cyclonedds_cpp

christophebedard, 2021-08-14:
There's a QD for Cyclone DDS itself, but there's no QD for rmw_cyclonedds_cpp. Shouldn't there be a QD for rmw_cyclonedds_cpp?

joespeed, 2021-08-14:
I assumed it is inclusive of rmw_cyclonedds but people that know best are @clalancette @ThijsSassen @k0ekk0ek @eboasson 
https://github.com/eclipse-cyclonedds/cyclonedds/blob/master/CYCLONEDDS_QUALITY_DECLARATION.md

christophebedard, 2021-08-14:
It does mention `rmw_cyclonedds` once, but it looks like it mostly applies to Cyclone DDS. It mentions `dds_*`/`DDS_*` symbols and links to stuff under https://github.com/eclipse-cyclonedds/cyclonedds.

I was mostly just wondering. I assume that a `rmw_cyclonedds_cpp`-specific quality declaration would simply be a formality (pointing to the existing Cyclone DDS QD as a direct runtime non-ROS dependency).

eboasson, 2021-08-16:
> I assume that a rmw_cyclonedds_cpp-specific quality declaration would simply be a formality (pointing to the existing Cyclone DDS QD as a direct runtime non-ROS dependency).

I suppose that aside from referring to Cyclone DDS's QD it would be a copy-paste from one of the core ROS packages as the RMW layer follows the ROS 2 process for reviews, CI, versioning, what-have-you. The only unusual, if perfectly logical, thing about the RMW layers is that tests are maintained separately.

clalancette, 2021-08-16:
In short, someone needs to go through [REP-2004](https://www.ros.org/reps/rep-2004.html) and accurately describe how it applies to `rmw_cyclonedds`.  Once that is done, we can come up with an accurate Quality Level for it.  My suggestion is to start with the template in https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_cpp/QUALITY_DECLARATION.md , and then change each section to fill it in as appropriate.

PR #335 (2021-09-13): Add quality declaration for rmw_cyclonedds_cpp
Closes #331

Looks like the current state of the package corresponds to QL 4.

It's technically supposed to be QL 1 for downstream packages (rcl, rclcpp, rclpy) to be QL 1, but some of the missing items are quite trivial (bumping to 1.0.0, creating missing upstream QDs).
closed 2021-09-13 by clalancette