ros2/rmw_cyclonedds #338: Galactic CycloneDDS crash

carteavcar01, 2021-09-22:
hey,
does anyone face with that log:
/tmp/binarydeb/ros-galactic-cyclonedds-0.8.0/src/core/ddsi/src/q_receive.c:3172: handle_submsg_sequence: Assertion `thread_is_asleep ()' failed

I've got a ground filter node, when one of the callback is called the node is crash and the log appear.

eboasson, 2021-09-22:
I know exactly what bug that is :( It is an off-by-one error in error handling, so where the two of the hardest problems in software come together ... it is in Cyclone 0.8.0rc1 and later. I'm hoping to get Galactic to switch from the rather special `iceoryx` pre-release branch to the official release now that it exists.

What is also interesting is that it means you received a malformed message. Unfortunately the `assert` is just before it prints that message so we don't know in what way the message was malformed ...

If you built from source, it is probably best to just use the release. If you want to just patch out this one problem, cherry-picking caae5482d3c40657451af48b4d11f6219bc2c3f7 will do the trick.

carteavcar01, 2021-09-23:
Thank you very much for the quick response!
can you give little bit more details about the patching please?

aprotyas, 2021-09-23:
> can you give little bit more details about the patching please?

If you've built Galactic from source, then you can do the following to patch:
```bash
cd path/to/ros_ws/src/eclipse-cyclonedds/cyclonedds
git fetch --all && git cherry-pick caae5482
```

eboasson, 2021-10-15:
@cottsay I can see why you would prefer that route if the next patch release is around the corner. So, yes, sure, go ahead!

cottsay, 2021-11-08:
@eboasson - we discussed this at a ROS 2 meeting, and cherry-picking the fix directly into the release repo won't affect source builds and nightly CI, and we decided that diverging the behavior wasn't a good idea.

What are the options for getting a fixed CycloneDDS for Galactic? Do you have available resources for testing a 0.8.x final release with Galactic, or do we need to bring the fix back to the `iceoryx` branch?

eboasson, 2021-11-09:
I think everyone would prefer to switch Galactic over to `0.8.1`. It should work out-of-the-box with Galactic, there'll be a deprecation warning from the compiler requiring a trivial fix (present in Humble already) but I assume that by:

> Do you have available resources for testing a 0.8.x final release with Galactic

you have some additional testing in mind. What exactly do you need?

eboasson, 2021-11-19:
@cottsay friendly ping, esp. also considering https://github.com/ros2/rmw_cyclonedds/pull/256

clalancette, 2023-06-13:
Given that Galactic is now EOL and this is presumably fixed in later distributions, I'm going to close this one out.

closed 2023-06-13 by clalancette