ros2/rmw_cyclonedds #339: User setting for Qos depth is ignored while using Clinet/Service

Barry-Xu-2018, 2021-09-28:
This problem was reported at ros2/rclcpp#1785.  

After checking code, find 

https://github.com/ros2/rmw_cyclonedds/blob/2aa2cf527d079265195b2dca91bb9c82481be0c1/rmw_cyclonedds_cpp/src/rmw_node.cpp#L4312

This line of code causes user settings to be ignored.  
Is there a reason for writing the code like this ? Or this is a bug ?

eboasson, 2021-09-29:
It is a bug (see https://github.com/ros2/rclcpp/issues/1785#issuecomment-929863731 for more info).

Barry-Xu-2018, 2021-09-29:
> It is a bug (see [ros2/rclcpp#1785 (comment)](https://github.com/ros2/rclcpp/issues/1785#issuecomment-929863731) for more info).

Thanks for your comments.   

Based on the fact that the user should be responsible for his actions, if the user doesn't configure, we can keep the current default setting.   
If the user wants to configure QoS, we accept user setting.