ros2/rmw_cyclonedds #354: Passing ROS2 services between RTI and CycloneDDS

eranroll, 2021-11-16:
Hi,

Is this option supported? From what I have tried in most cases the services will not return the response, or will never become available. Is that supported right now in ROS Galactic versions and RTI Connext 6.0.1?

If it is supported, is there any specific DDS xml configuration that can make it work, which isn't the default config?

Regards,
Eran

eranroll, 2021-11-16:
Attached is a log of CycloneDDS (finest) when the server is the Cyclone Node, and the client (who asked the service) is RTI
[service_rti_debug.txt](https://github.com/ros2/rmw_cyclonedds/files/7549343/service_rti_debug.txt)

clalancette, 2021-11-16:
As far as I understand it, no, service calls between different DDS vendors is not guaranteed to work.  That's because the way that services are implemented is not standardized in DDS, so the different vendors may do it differently.

clalancette, 2022-02-04:
Given that this was a question, and we are tracking service interoperability between DDS vendors elsewhere, I'm going to close this one out.

closed 2022-02-04 by clalancette