ros2/rmw_cyclonedds #362: Incompatible cyclonedds breaks with 0.22.4 galactic backport

Aposhian, 2022-01-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - 0.22.4
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
docker run -it ros:galactic@sha256:e88918e911975edd2d2ae88c1078561146665fdb1a836d044b132913f1a7cbf7
apt-get update && apt-get install -y ros-galactic-rmw-cyclonedds-cpp
nm -gD /opt/ros/galactic/lib/librmw_cyclonedds_cpp.so  | grep loan_available
nm -gD /opt/ros/galactic/lib/x86_64-linux-gnu/libddsc.so.0 | grep loan_available
```

#### Expected behavior
`librmw_cyclonedds_cpp` should have a reference to `dds_is_loan_available` and so should `libddsc`.

#### Actual behavior
`librmw_cyclonedds_cpp` has a reference to `dds_is_loan_available` but `libddsc` only has `is_loan_available`. This is because the debian package for `ros-galactic-cyclonedds` was not updated along with `ros-galactic-rmw-cyclonedds-cpp` since the debian package dependency is _unversioned_ as seen here:
```
root@61f7f94841c0:/# apt depends ros-galactic-rmw-cyclonedds-cpp
ros-galactic-rmw-cyclonedds-cpp
  Depends: libc6 (>= 2.14)
  Depends: libgcc-s1 (>= 3.0)
  Depends: libstdc++6 (>= 9)
  Depends: ros-galactic-cyclonedds
  Depends: ros-galactic-rcpputils
  Depends: ros-galactic-rcutils
  Depends: ros-galactic-rmw
  Depends: ros-galactic-rmw-dds-common
  Depends: ros-galactic-rosidl-runtime-c
  Depends: ros-galactic-rosidl-typesupport-introspection-c
  Depends: ros-galactic-rosidl-typesupport-introspection-cpp
  Depends: ros-galactic-ros-workspace
```

#### Additional information
While this could be solved by simply adding a `version_gte` attribute to the `<depends>cyclonedds</depends>` tag in the `package.xml` for `rmw_cyclonedds_cpp`, the debian package versions for `ros-*-cyclonedds` are not managed very well, so we would have to use build numbers to distinguish between `0.8.0-5focal.20210608.002038` and `0.8.0-6focal.20211217.210911`. So

Aposhian, 2022-01-03:
Caused by: https://github.com/ros2/rmw_cyclonedds/pull/336

clalancette, 2022-01-04:
> `librmw_cyclonedds_cpp` has a reference to `dds_is_loan_available` but `libddsc` only has `is_loan_available`. This is because the debian package for `ros-galactic-cyclonedds` was not updated along with `ros-galactic-rmw-cyclonedds-cpp`

That doesn't seem to be the case: https://github.com/ros/rosdistro/pull/31510 and https://github.com/ros/rosdistro/pull/31555 were both merged for the latest sync.  Indeed, if I run your test on the packaged versions I have:

```
$ nm -gD /opt/ros/galactic/lib/librmw_cyclonedds_cpp.so  | grep loan_available
                 U dds_is_loan_available
$ nm -gD /opt/ros/galactic/lib/x86_64-linux-gnu/libddsc.so.0 | grep loan_available
00000000000c4260 T dds_is_loan_available
00000000000c42f0 T is_loan_available
```

My version of the packages is:
```
ubuntu@ros2-focal:~$ dpkg -l | egrep '.*galactic.*cyclonedds'
ii  ros-galactic-cyclonedds                             0.8.0-6focal.20211217.210911         amd64        Eclipse Cyclone DDS is a very performant and robust open-source DDS implementation.
ii  ros-galactic-rmw-cyclonedds-cpp                     0.22.4-1focal.20211222.231803        amd64        Implement the ROS middleware interface using Eclipse CycloneDDS in C++.
```

What versions do you have?

Aposhian, 2022-01-04:
I have the following versions (in the latest build of the `ros:galactic` container):
```
$ docker run ros:galactic bash -c "dpkg -l | grep cyclone"
ii  ros-galactic-cyclonedds                             0.8.0-5focal.20210608.002038         amd64        Eclipse Cyclone DDS is a very performant and robust open-source DDS implementation.
ii  ros-galactic-rmw-cyclonedds-cpp                     0.22.3-1focal.20210805.162212        amd64        Implement the ROS middleware interface using Eclipse CycloneDDS in C++.
```
but then I manually do
```
apt update && apt install ros-galactic-rmw-cyclonedds-cpp
```
which brings me to this state, which is broken:
```
ii  ros-galactic-cyclonedds                             0.8.0-5focal.20210608.002038         amd64        Eclipse Cyclone DDS is a very performant and robust open-source DDS implementation.
ii  ros-galactic-rmw-cyclonedds-cpp                     0.22.4-1focal.20211222.231803        amd64        Implement the ROS middleware interface using Eclipse CycloneDDS in C++.
```

So `rmw_cyclonedds_cpp` 0.22.4 actually depends on having `cyclonedds` 0.8.0-6 or greater, but since the apt package dependency is unversioned, it's perfectly happy to leave it at 0.8.0-5, which leads to the problem.

clalancette, 2022-01-04:
Ah, I see what you mean, thanks for the clarification..  Since the package is already installed in your case, upgrading just `rmw-cyclonedds-cpp` causes it to fail.  We don't generally recommend doing that (for exactly this reason), but it is a bit unfortunate.

One thing we could do is to add a patch to the release repository to make sure `rmw_cyclonedds_cpp` does depend on `cyclonedds` 0.8.0-6 or later.  But I'm not an expert here, so I'll defer to @nuclearsandwich and @cottsay for thoughts.

eboasson, 2022-01-05:
@clalancette @nuclearsandwich @cottsay the root cause is that I forgot to update `package.xml` when bumping version numbers. I need to add a check in the CI because I am bound to make the same mistake again.

Aposhian, 2022-01-05:
Once we have a cyclonedds Debian package with the version number bumped to 0.8.2, then this should be resolved by adding `version_gte=0.8.2` to the `<depends>cyclonedds</depends>` tag, which will allow Bloom to propagate that requirement to the Debian package itself.

eboasson, 2022-01-10:
Cyclone now has a 0.8.2 where the package.xml version is correct.

Aposhian, 2022-01-10:
It looks like the `ros-galactic-cyclonedds` 0.8.2 package hasn't made it to the apt repo yet, but I have made a draft PR that adds the version dependency for when it does. And as a reminder for everyone: the version tagging doesn't help you if you are building from source and using rosdep to install dependencies: it just helps when you are installing both `cyclonedds` and `rmw_cyclonedds` as debian packages.

alexleel, 2023-04-01:
> $ nm -gD /opt/ros/galactic/lib/librmw_cyclonedds_cpp.so  | grep loan_available
>                  U dds_is_loan_available
> $ nm -gD /opt/ros/galactic/lib/x86_64-linux-gnu/libddsc.so.0 | grep loan_available
> 00000000000c4260 T dds_is_loan_available
> 00000000000c42f0 T is_loan_available

Hi  I install cyclonedds as https://github.com/eclipse-cyclonedds/cyclonedds
However. the libddsc.so.0  does not have dds_is_loan_available as below:
$nm -gD libddsc.so.0 | grep loan_available
00000000000febdb T is_loan_available
And librmw_cyclonedds_cpp has this reference , so when I run some test as 
$ros2 run ros2_shm_demo talker
ros2_shm_demo/lib/ros2_shm_demo/talker: symbol lookup error: /opt/ros/galactic/lib/librmw_cyclonedds_cpp.so: undefined symbol: dds_is_loan_available

What can I do to make the version compatible?

Thanks.