ros2/rmw_cyclonedds #387: Using mv_cyclonedds , not working comunication between two computers in the same network

lunanueva, 2022-04-08:
- DDS implementation:
  - mv_cyclonedds
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior
Using talker / listener, nodes, in different computer  with sucessfull result
#### Actual behavior
Not messages be received in the listener node
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

eboasson, 2022-04-19:
It would help if you could provide some additional information. I did notice that you copy-pasted some fragment of a Cyclone DDS trace in the description, but (curiously) that fragment suggests there *is* communication:
```
1649380237.027044 [9] dq.builtin: SPDP ST0 2baf1001:f7a89bf:8cf2f5b9:1c1 bes c3f NEW (0x00000000-0x0000002c-0x00000000-0x00000000-0x00000000 mycompu/0.7.0/Linux/Linux) (data udp/239.255.0.1:9651 udp/192.168.1.71:43442 meta udp/239.255.0.1:9650 udp/192.168.1.71:43442) QOS={user_data=10<"enclave=/;">}
```
is the discovery of some participant in another process, but I can't tell if it is on the same machine or a different machine (because I don't know the IP address of the machine running the process that produced this trace fragment).

clalancette, 2023-06-13:
No response in over a year, so closing this out.  If you are still having problems, please open a question on https://answers.ros.org, our central question and answer site.

closed 2023-06-13 by clalancette