ros2/rmw_cyclonedds #397: Content Filtering Topics Support

eranroll, 2022-06-24:
Hi,

This is somewhat of a feature request, rather than an issue.
We have been working with CycloneDDS as our preferred ROS2 RMW in Galactic Distro for quite sometime, and the performance has been great!

However, reading about ROS2 Humble's roadmap, I see that content filtering topics are not supported with your RMW.
Is that a support you are planning on having within Humble?

Regards

eboasson, 2022-07-18:
It is a bit of a complicated answer, because there is what Cyclone DDS will do natively and what can be done today in the RMW layer.

The RMW layer uses a custom sample representation/(de)serialiser and can actually implement subscriber-side filtering today by installing a filter predicate (a regular C function taking a sample and returning a bool). ROS 2 provides all the metadata already. Obviously, there's no way a function pointer can be transferred to the publisher, and so this is inherently limited in what it can do.

The native Cyclone DDS answer is that we want "proper" content filtering, where the application passes an expression instead of a function, and then we can transfer that expression to the publishers and do the filtering on the publisher side.(*) For this, we've been waiting for sufficient metadata to be available, that took a bit longer than expected because the DDS XTypes spec that defines interoperable metadata is such a horror-show. But that's all been sorted now, and we've been debating what would be the most elegant way to actually implement this.

We support multiple language bindings, serialization formats, data representations, &c., and so all this needs to be parameterized. The RMW layer can then use the same mechanisms to make it work properly.

The intention is definitely to have this operational within the Humble Year. Whether it'll become standard in Humble or whether you'd need to build some parts from source is an open question. Open Robotics works hard to keep a given distribution stable, and it depends on the magnitude of the changes and the attendant risk how much testing would be involved in doing a significant update of Cyclone DDS within Humble's life cycle.

(*) There is already such a thing as publisher-side filtering, in fact, DDS Security relies on it quite heavily. The big difference is that with DDS Security, we know the types and we know the expressions, and so there is no problem having a local function definition that evaluates the predicate.

christomaszewski, 2024-04-30:
Has content filtering been implemented within rmw-cyclonedds yet? Is there perhaps an alternative to achieve the same network traffic reduction with zenoh?

eboasson, 2024-06-18:
> Has content filtering been implemented within rmw-cyclonedds yet?

No, I'm afraid not.

> Is there perhaps an alternative to achieve the same network traffic reduction with zenoh?

A lot of interesting used to be possible by using the Zenoh-DDS bridge, but now Zenoh has its [very own RMW layer](https://github.com/ros2/rmw_zenoh) and it is definitely worth a look. I don't know whether it implements content filtering, though.