ros2/rmw_cyclonedds #400: rmw_cyclonedds_cpp: Cannot locate rosdep definition for [iceoryx_binding_c]

qwe789qwec, 2022-07-18:
```
rosdep install --from-paths ~/ros2_dashing/ros2-linux/share --ignore-src --rosdistro dashing -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 osrf_testing_tools_cpp poco_vendor rmw_connext_cpp rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rti-connext-dds-5.3.1 tinyxml_vendor tinyxml2_vendor urdfdom urdfdom_headers"
WARNING: given --rosdistro dashing but ROS_DISTRO is "humble". Ignoring environment.
apt-cache: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/wilson/ros2_dashing/ros2-linux/lib/libzstd.so.1)
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rmw_cyclonedds_cpp: Cannot locate rosdep definition for [iceoryx_binding_c]
```

Tried a lot of methods can not be solved.

already try rosdep update --include-eol-distros
Directly downloading from git and putting it into src doesn't solve it either

eboasson, 2022-07-18:
Dashing was a very long time ago, I'm pretty sure `iceoryx_binding_c` didn't exist at the time and there definitely was no Cyclone+Iceoryx integration back then: it only became official in Galactic, and so I think Galactic would be the first distribution to have a rosdep definition for `iceoryx_binding_c`.

Why in Dashing `rosdep` would try to be solve some sort of a dependency on `iceoryx_binding_c` is a mystery to me, too. The `package.xml` file for Dashing/Eloquent doesn't mention anything Iceoryx.

qwe789qwec, 2022-07-18:
I try to install ros2 by this step

https://docs.ros.org/en/dashing/Installation/Ubuntu-Install-Binary.html#installing-the-missing-dependencies

But I'm stuck on "installing the missing dependencies".

clalancette, 2022-07-26:
Yeah, dashing is way out-of-date and not supported anymore.  I'll suggest trying with a newer distribution; Humble is the latest, and the installation instructions for it are at https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html

closed 2022-07-26 by clalancette