ros2/rmw_cyclonedds #401: Shared memory not working with transient_local durability

afrixs, 2022-08-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.1
- Installation type:
  - binaries
- Version or commit hash:
  - 1.4.0-1jammy.20220711.152242
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Run robot_state_publisher which publishes to /tf_static using transient_local durability and afterwards create a subscription for the topic.
```
#Terminal 1
iox-roudi

#Terminal 2
CYCLONEDDS_URI="<CycloneDDS><Domain id='any'><SharedMemory><Enable>true</Enable></SharedMemory></Domain></CycloneDDS>" ros2 run robot_state_publisher robot_state_publisher --ros-args -p robot_description:="<robot name='test'><link name='base_link'/><link name='imu_link'/><joint name='base_link_to_imu_link' type='fixed'><parent link='base_link'/><child link='imu_link'/></joint></robot>"

#Terminal 3
CYCLONEDDS_URI="<CycloneDDS><Domain id='any'><SharedMemory><Enable>true</Enable></SharedMemory></Domain></CycloneDDS>" ros2 topic echo /tf_static --qos-durability transient_local --qos-reliability reliable --qos-depth 1
```

#### Expected behavior
A message is received, although echo is called later than the message has been published, because that's what transient_local should take care of.

#### Actual behavior
No message is received. If Terminal 2 command is launched after Terminal 3 command, the message gets received, but that's not sufficient.

#### Additional information
When called without shared memory, i.e.
```
#Terminal 1
CYCLONEDDS_URI="<CycloneDDS/>" ros2 run robot_state_publisher robot_state_publisher --ros-args -p robot_description:="<robot name='test'><link name='base_link'/><link name='imu_link'/><joint name='base_link_to_imu_link' type='fixed'><parent link='base_link'/><child link='imu_link'/></joint></robot>"

#Terminal 2
CYCLONEDDS_URI="<CycloneDDS/>" ros2 topic echo /tf_static --qos-durability transient_local --qos-reliability reliable --qos-depth 1
```
, everything works as expected.

----

llapx, 2022-08-31:
Can not reproduct with latest ros:rolling, can you provide more detail log?

afrixs, 2022-08-31:
Package versions:
```
$ dpkg -l | grep -E "cyclonedds|iceoryx|ros2cli|ros-rolling-robot-state-publisher"
ii  ros-rolling-cyclonedds                             0.9.0-1jammy.20220711.134212                                amd64        Eclipse Cyclone DDS is a very performant and robust open-source DDS implementation.
ii  ros-rolling-iceoryx-binding-c                      2.0.2-1jammy.20220711.133552                                amd64        Eclipse iceoryx inter-process-communication (IPC) middleware C-Language Binding
ii  ros-rolling-iceoryx-hoofs                          2.0.2-1jammy.20220711.130813                                amd64        Eclipse iceoryx inter-process-communication (IPC) middleware basic building blocks
ii  ros-rolling-iceoryx-posh                           2.0.2-1jammy.20220711.131117                                amd64        Eclipse iceoryx inter-process-communication (IPC) middleware Posix Shared Memory Library and middleware daemon (RouDi)
ii  ros-rolling-rmw-cyclonedds-cpp                     1.4.0-1jammy.20220726.201039                                amd64        Implement the ROS middleware interface using Eclipse CycloneDDS in C++.
ii  ros-rolling-rmw-cyclonedds-cpp-dbgsym              1.4.0-1jammy.20220726.201039                                amd64        debug symbols for ros-rolling-rmw-cyclonedds-cpp
ii  ros-rolling-robot-state-publisher                  3.1.0-1jammy.20220815.144819                                amd64        This package take the joint angles of a robot as input, and publishes the 3D poses of the robot links to tf2, using a kinematic tree model of the robot.
ii  ros-rolling-ros2cli                                0.19.0-1jammy.20220726.233741                               amd64        Framework for ROS 2 command line tools.
ii  ros-rolling-ros2cli-common-extensions              0.2.0-1jammy.20220727.035404                                amd64        Meta package for ros2cli common extensions

$ iox-roudi -v
RouDi version: 2.0.2
Build date: 2022-07-11T13:11:18Z
Commit ID:
```

`iox-roudi` log (+ added separating lines after each step):
```
$ iox-roudi -l verbose
2022-08-31 10:02:45.197 [ Info  ]: No config file provided and also not found at '/etc/iceoryx/roudi_config.toml'. Falling back to built-in config.
Log level set to: [Verbose]
2022-08-31 10:02:45.197 [Verbose]: Command line parameters are:
Log level: Verbose
Monitoring mode: MonitoringMode::ON
Compatibility check level: CompatibilityCheckLevel::PATCH
Unique RouDi ID: < unset >
Process kill delay: 45 s
Config file used is: < none >
Reserving 66761736 bytes in the shared memory [iceoryx_mgmt]
[ Reserving shared memory successful ]
2022-08-31 10:02:45.233 [ Debug ]: Registered memory segment 0x7ffbc8d72000 with size 66761736 to id 1
Reserving 149264720 bytes in the shared memory [matej]
[ Reserving shared memory successful ]
2022-08-31 10:02:45.312 [ Debug ]: Roudi registered payload data segment 0x7ffbbff18000 with size 149264720 to id 2
RouDi is ready for clients

######################## after running Terminal1 command ############################

2022-08-31 10:03:16.076 [ Debug ]: Registered new application iceoryx_rt_134727_1661932996075633044
2022-08-31 10:03:16.077 [ Debug ]: Created new ConditionVariable for application iceoryx_rt_134727_1661932996075633044
2022-08-31 10:03:16.079 [ Debug ]: Created new PublisherPort for application 'iceoryx_rt_134727_1661932996075633044' with service description 'Service: DDS_CYCLONE, Instance: rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_, Event: ros_discovery_info'
2022-08-31 10:03:16.093 [ Debug ]: Created new PublisherPort for application 'iceoryx_rt_134727_1661932996075633044' with service description 'Service: DDS_CYCLONE, Instance: rcl_interfaces::srv::dds_::GetParameters_Response_, Event: rr/robot_state_publisher/get_parametersReply'
2022-08-31 10:03:16.103 [ Debug ]: capro::SUB/UNSUB, no matching publisher for subscriber from runtime 'iceoryx_rt_134727_1661932996075633044' and with service description 'Service: DDS_CYCLONE, Instance: rcl_interfaces::srv::dds_::GetParameters_Request_, Event: rq/robot_state_publisher/get_parametersRequest'!
2022-08-31 10:03:16.103 [ Debug ]: Created new SubscriberPort for application 'iceoryx_rt_134727_1661932996075633044' with service description 'Service: DDS_CYCLONE, Instance: rcl_interfaces::srv::dds_::GetParameters_Request_, Event: rq/robot_state_publisher/get_parametersRequest'
2022-08-31 10:03:16.104 [ Debug ]: Created new PublisherPort for application 'iceoryx_rt_134727_1661932996075633044' with service description 'Service: DDS_CYCLONE, Instance: rcl_interfaces::srv::dds_::GetParameterTypes_Response_, Event: rr/robot_state_publisher/get_parameter_typesReply'
2022-08-31 10:03:16.115 [ Debug ]: capro::SUB/UNSUB, no matching publisher for subscriber from runtime 'iceoryx_rt_134727_1661932996075633044' and with service description 'Service: DDS_CYCLONE, Instance: rcl_interfaces::srv::dds_::GetParameterTypes_Request_, Event: rq/robot_state_publisher/get_parameter_typesRequest'!
2022-08-31 10:03:16.115 [ Debug ]: Created new SubscriberPort for application 'iceoryx_rt_134727_1661932996075633044' with service description 'Service: DDS_CYCLONE, Instance: rcl_interfaces::srv::dds_::GetParameterTypes_Request_, Event: rq/robot_state_publisher/get_parameter_typesRequest'
[... cut ...]
2022-08-31 10:05:21.298 [ Debug ]: Created new SubscriberPort for application 'iceoryx_rt_134862_1661933121138739923' with service description 'Service: DDS_CYCLONE, Instance: sensor_msgs::msg::dds_::JointState_, Event: rt/joint_states'
2022-08-31 10:05:21.299 [ Debug ]: Created new SubscriberPort for application 'iceoryx_rt_134862_1661933121138739923' with service description 'Service: DDS_CYCLONE, Instance: rcl_interfaces::msg::dds_::ParameterEvent_, Event: rt/parameter_events'

################################## after re-running Terminal2 command (a message got received) ###################

```

llapx, 2022-08-31:
@afrixs 

sorry, I forget to `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`, now I can reproduct the BUG.

afrixs, 2022-08-31:
Edited the commands so that this doesn't happen to anyone else

bopeng-saitama, 2023-02-13:
@afrixs 
I also encountered the same issue in my environment. Have you solved the problem?
I have made a  reproduce repo for this issue. https://github.com/bopeng-saitama/transient_local_iceoryx
Steps to reproduce issue:
```
#Terminal 1
iox-roudi

#Terminal 2
cd ~/transient_local_iceoryx
source /opt/ros/humble/setup.bash
source install/setup.bash
CYCLONEDDS_URI=file://$PWD/cyclonedds.xml ros2 run transient_local_iceoryx transient_local_pub

#Terminal 3
cd ~/transient_local_iceoryx
source /opt/ros/humble/setup.bash
source install/setup.bash
CYCLONEDDS_URI=file://$PWD/cyclonedds.xml ros2 run transient_local_iceoryx transient_local_sub
```
And according to my investigation, although the message published by pub was stored in the history, when sub was started, the operation of taking it out from the history and into the queue was not executed.(The corresponding function is `tryAddQueue()` which is defined in `iceoryx_posh/include/iceoryx_posh/internal/popo/building_blocks/chunk_distributor.inl`)

eboasson, 2023-02-14:
The other day I discovered what appeared to be a problem in Iceoryx' handling of publisher histories, where it would receive the data in the Iceoryx subscription but not notify the "application" — here, that would be Cyclone — that there is data ready for reading. Once fresh data arrives, it triggers and suddenly everything shows up.

I'm wondering if this matches the symptom you are observing. I'm not sure whether there is fresh data being published in your case.

Also I haven't reported it to the Iceoryx guys yet.

bopeng-saitama, 2023-02-14:
Thanks for the reply. 
>  I'm wondering if this matches the symptom you are observing. 

Yes, the symptom is the same as waht I observed.

afrixs, 2023-02-15:
> @afrixs I also encountered the same issue in my environment. Have you solved the problem?

Unfortunately I haven't... Currently I run nodes which need to publish transient_local topics without shared memory, that's my temporary "workaround"

duyu, 2024-07-05:
I ran into this issue as well, is there any update on this after another 1 year?

```
$ dpkg -l | grep -E "cyclonedds|iceoryx|ros2cli|robot-state-publisher"
ii  ros-humble-cyclonedds                             0.10.4-1jammy.20240217.025716           amd64        Eclipse Cyclone DDS is a very performant and robust open-source DDS implementation.
ii  ros-humble-iceoryx-binding-c                      2.0.5-1jammy.20240217.025528            amd64        Eclipse iceoryx inter-process-communication (IPC) middleware C-Language Binding
ii  ros-humble-iceoryx-hoofs                          2.0.5-1jammy.20240217.024457            amd64        Eclipse iceoryx inter-process-communication (IPC) middleware basic building blocks
ii  ros-humble-iceoryx-posh                           2.0.5-1jammy.20240217.024825            amd64        Eclipse iceoryx inter-process-communication (IPC) middleware Posix Shared Memory Library and middleware daemon (RouDi)
ii  ros-humble-rmw-cyclonedds-cpp                     1.3.4-1jammy.20240217.062033            amd64        Implement the ROS middleware interface using Eclipse CycloneDDS in C++.
ii  ros-humble-robot-state-publisher                  3.0.3-2jammy.20240217.085950            amd64        This package take the joint angles of a robot as input, and publishes the 3D poses of the robot links to tf2, using a kinematic tree model of the robot.
ii  ros-humble-ros2cli                                0.18.9-1jammy.20240217.070501           amd64        Framework for ROS 2 command line tools.
ii  ros-humble-ros2cli-common-extensions              0.1.1-4jammy.20240217.081520            amd64        Meta package for ros2cli common extensions
```

```
$ iox-roudi -v
RouDi version: 2.0.5
Build date: 2024-02-17T02:48:26Z
Commit ID: 
```

mcp-ats, 2025-04-29:
OK, I went through this issue and the one reported in IOX.

The latter has been closed stating that the issue lies in the upper framework, therefore I would say RMW.

Is anyone else using CycloneDDS with SHM and transient_local or is everyone avoiding transient_local and moving this to a non SHM enabled process?