ros2/rmw_cyclonedds #403: fatal error: iceoryx_binding_c/chunk.h: No such file or directory

king318hua, 2022-08-16:
## Bug report

when building ros2  galactic

 colcon build --symlink-install
-----------------------------------------------------------------

--- stderr: rmw_cyclonedds_cpp
In file included from /home/kw/ros2_galactic/install/cyclonedds/include/dds/ddsi/q_xmsg.h:24,
                 from /home/kw/ros2_galactic/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/serdata.hpp:26,
                 from /home/kw/ros2_galactic/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/serdata.cpp:14:
/home/kw/ros2_galactic/install/cyclonedds/include/dds/ddsi/ddsi_shm_transport.h:24:10: fatal error: iceoryx_binding_c/chunk.h: No such file or directory
   24 | #include "iceoryx_binding_c/chunk.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rmw_cyclonedds_cpp.dir/build.make:89: CMakeFiles/rmw_cyclonedds_cpp.dir/src/serdata.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/kw/ros2_galactic/install/cyclonedds/include/dds/ddsi/q_xmsg.h:24,
                 from /home/kw/ros2_galactic/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/serdata.hpp:26,
                 from /home/kw/ros2_galactic/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/Serialization.hpp:21,
                 from /home/kw/ros2_galactic/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:55:
/home/kw/ros2_galactic/install/cyclonedds/include/dds/ddsi/ddsi_shm_transport.h:24:10: fatal error: iceoryx_binding_c/chunk.h: No such file or directory
   24 | #include "iceoryx_binding_c/chunk.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rmw_cyclonedds_cpp.dir/build.make:76: CMakeFiles/rmw_cyclonedds_cpp.dir/src/rmw_node.cpp.o] Error 1
In file included from /home/kw/ros2_galactic/install/cyclonedds/include/dds/ddsi/q_xmsg.h:24,
                 from /home/kw/ros2_galactic/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/serdata.hpp:26,
                 from /home/kw/ros2_galactic/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/Serialization.hpp:21,
                 from /home/kw/ros2_galactic/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/Serialization.cpp:21:
/home/kw/ros2_galactic/install/cyclonedds/include/dds/ddsi/ddsi_shm_transport.h:24:10: fatal error: iceoryx_binding_c/chunk.h: No such file or directory
   24 | #include "iceoryx_binding_c/chunk.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2022-09-08:
So I just tried to build Galactic from source up to `rmw_cyclonedds_cpp`, and it seemed to compile fine for me.  Can you give more details on how you checked out the sources, and the exact commands you used?

king318hua, 2022-10-14:
> So I just tried to build Galactic from source up to `rmw_cyclonedds_cpp`, and it seemed to compile fine for me. Can you give more details on how you checked out the sources, and the exact commands you used?
i'm sorry,its too long that i have forgotten the details.
the file "https://github.com/eclipse-iceoryx/iceoryx/blob/v2.0.2/iceoryx_posh/CMakeLists.txt "(line 47~60):
```
option(MAKE_UNIQUE_INCLUDEDIR
  "When ON headers are installed to a path ending with folders called \
  iceoryx/vX.Y.Z/ . This avoids include directory search order issues when \
  overriding this package from a merged catkin, ament, or colcon workspace."
  ON)

if(MAKE_UNIQUE_INCLUDEDIR)
  set(_unique_dir "iceoryx/v${IOX_VERSION_STRING}")
  if(PREFIX STREQUAL "")
    set(PREFIX "${_unique_dir}")
  else()
    set(PREFIX "${PREFIX}/${_unique_dir}")
  endif()
endif()
"
```