ros2/rmw_cyclonedds #409: Include topic names in error messages

tfoote, 2022-08-30:
## Feature request

#### Feature description
We added better error messages when topics are invalid in #405 but the topics aren't included in the error message. It would be great to include them. 

#### Implementation considerations

The topic name can be passed into this function to be included inthe error strings. 

https://github.com/ros2/rmw_cyclonedds/pull/405/files#diff-91f31a99b38624440900850fd2df4710f2ef46c140999f3e3b8b89393e48e202R1790

And the name passed in in the 4 places it's invoked:

https://github.com/ros2/rmw_cyclonedds/pull/405/files#diff-91f31a99b38624440900850fd2df4710f2ef46c140999f3e3b8b89393e48e202R2296

https://github.com/ros2/rmw_cyclonedds/pull/405/files#diff-91f31a99b38624440900850fd2df4710f2ef46c140999f3e3b8b89393e48e202R2806

https://github.com/ros2/rmw_cyclonedds/pull/405/files#diff-91f31a99b38624440900850fd2df4710f2ef46c140999f3e3b8b89393e48e202R4638

https://github.com/ros2/rmw_cyclonedds/pull/405/files#diff-91f31a99b38624440900850fd2df4710f2ef46c140999f3e3b8b89393e48e202R4647

tfoote, 2022-08-30:
@Voldivh FYI

Voldivh, 2022-08-31:
Hey @tfoote, take a look at the PR #410 and let me know.

clalancette, 2022-09-29:
We completed this in #410, so closing this issue.

closed 2022-09-29 by clalancette