ros2/rmw_cyclonedds #412: Action based CI Broken for foxy pull requests

tfoote, 2022-08-31:
## Bug report

**Required Info:**

- Operating System:
  - All
- Installation type:
  - CI
- Version or commit hash:
  - Foxy
- DDS implementation:
  - Default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Open a PR against the foxy branch with no or trivial changes. 

#### Expected behavior

CI Passes

#### Actual behavior

CI Fails

#### Additional information

I found this when submitting: https://github.com/ros2/rmw_cyclonedds/pull/408

I've tried to fix two different CI issues in: https://github.com/ros2/rmw_cyclonedds/pull/411

christophebedard, 2022-08-31:
Except for the Ubuntu-based jobs, the GitHub workflows have never really worked/passed on this repo.

tfoote, 2022-08-31:
Looking at this as someone who's not usually contributing here, I've apparently wasted a lot of time trying to get them to work. Should we just remove them so others don't have the same problem?

christophebedard, 2022-08-31:
That's probably a good idea.

cc @clalancette @jacobperron @sloretz @eboasson since you all seem to be the usual contributors.

clalancette, 2022-08-31:
> cc @clalancette @jacobperron @sloretz @eboasson since you all seem to be the usual contributors.

That's fine by me, they've never worked well here.

christophebedard, 2022-09-01:
I can take care of this. #372 is already a good first step. Should we keep the Ubuntu jobs?

clalancette, 2022-09-01:
> I can take care of this. #372 is already a good first step. Should we keep the Ubuntu jobs, or just remove everything?

I'm inclined to remove everything (we already have the Rpr jobs, and CI is required for everything anyway), but I don't feel strongly about it.

jacobperron, 2022-09-01:
Since it doesn't appear anyone is maintaining the GitHub workflows, I would remove them entirely.

jacobperron, 2022-09-01:
GitHub CI jobs removed in #414 for Rolling.

Here are backports for other distros:

- https://github.com/ros2/rmw_cyclonedds/pull/415
- https://github.com/ros2/rmw_cyclonedds/pull/416
- https://github.com/ros2/rmw_cyclonedds/pull/417

closed 2022-09-02 by jacobperron