ros2/rmw_cyclonedds #443: ROS2 examples_rclcpp_minimal_service test crash

zhukao, 2023-03-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - ROS2 Foxy, ros-foxy-rmw-cyclonedds-cpp/focal,now 0.7.11-1focal.20221012.222818 arm64
- DDS implementation:
  - cyclonedds
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

terminal 1
```
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
ros2 run examples_rclcpp_minimal_service service_main
```
terminal 2, run `bash test.sh`
```
# cat test.sh
#!/bin/bash

for i in {1..35}
do
    echo $i
    ros2 run examples_rclcpp_minimal_client client_main --ros-args -r __node:=my_client_$i &
done
```
```
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
bash test.sh
```

#### Expected behavior
terminal 1 outputs '[minimal_service]: request: 41 + 1'.
terminal 2 outputs '[minimal_client]: result of 41 + 1 = 42'.

#### Actual behavior
terminal 1 crashes with outpus
```
[INFO] [1679455265.326423020] [minimal_service]: request: 41 + 1
[INFO] [1679455265.381029282] [minimal_service]: request: 41 + 1
[INFO] [1679455265.594560479] [minimal_service]: request: 41 + 1
[INFO] [1679455265.636475358] [minimal_service]: request: 41 + 1
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to send response: cannot publish data, at /tmp/binarydeb/ros-foxy-rmw-cyclonedds-cpp-0.7.11/src/rmw_ns-foxy-rcl-1.1.14/src/rcl/service.c:356
```
terminal 2 outpus
```
[INFO] [1679455265.190357743] [my_client_10]: result of 41 + 1 = 42
[INFO] [1679455265.372295787] [my_client_25]: result of 41 + 1 = 42
[INFO] [1679455265.634295150] [my_client_33]: result of 41 + 1 = 42
[INFO] [1679455265.677879464] [my_client_24]: result of 41 + 1 = 42
[INFO] [1679455266.310747479] [my_client_30]: result of 41 + 1 = 42
[INFO] [1679455268.346233987] [my_client_3]: result of 41 + 1 = 42
[INFO] [1679455268.732088645] [my_client_26]: waiting for service to appear...
[INFO] [1679455269.091401288] [my_client_35]: waiting for service to appear...
[INFO] [1679455269.100591599] [my_client_20]: waiting for service to appear...
[INFO] [1679455269.732822395] [my_client_26]: waiting for service to appear...
[INFO] [1679455270.092682560] [my_client_35]: waiting for service to appear...
```

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

zhukao, 2023-03-22:
The crash dose not appear steadily, run `bash test.sh` about 5 times the service may appear crash.

clalancette, 2023-03-30:
Hi there, thanks for reporting the bug.  Using the script you provided, I was not able to reproduce this error on either Foxy or Rolling, even after running a number of times.  Could you give some more information about your environment (running in a container, etc)?  Thanks.

zhukao, 2023-03-31:
Thank you for your reply. I am running on an actual machine, not a container. The installation packages for arm64 and amd64 can all be reproduced. The reproduction time will vary depending on the device, but they will all be reproduced.
You can try using a script to loop and test the client, eg `bash loop.sh`, and the script is: 
```
# cat loop.sh
#!/bin/bash

for i in {1..100}
do
    echo -e "\n loop " $i
    bash test.sh
    sleep 30
done
````