ros2/rmw_cyclonedds #452: Memory leak slowly and chunk size increase with multiple nodes

Lannister-Xiaolin, 2023-04-20:
## Bug report

**Required Info:**

- Operating System:
Ubuntu20.04 Arm64
  - 
- Installation type:
- binaries 
- DDS implementation:
  - 
  cyclonedds
- Client library (if applicable):
  - 
rclcpp

#### Steps to reproduce issue
There there nodes, one is for image node witch publish fixed size image by zero copy mode
Then there two nodes(name: NodeA and Node B) subscribe to the fixed image,  and publish some point cloud data(without zero copy mode)
For these nodes, we set env variables RMW_IMPLEMENTATION and CYCLONEDDS_URI correctly
Then, when I echo point cloud published by (NodeA and Node B) in the terminal without setting RMW_IMPLEMENTATION and CYCLONEDDS_URI.  We can see memoy increase
[image]

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2023-04-20:
Which ROS 2 release are you using?

Lannister-Xiaolin, 2023-04-21:
> Which ROS 2 release are you using?

Galactic

clalancette, 2023-04-21:
> Galactic

Can you try with Humble and see if the situation is any better?  There were a few upgrades to the iceoryx integration in that release.  Note that if you want to stick with Ubuntu 20.04, you'll have to build Humble from source: http://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html

tdinesh, 2023-11-16:
#365 solves this for rolling. 

For galactic I have a backport at [galactic 365](https://github.com/tdinesh/rmw_cyclonedds/tree/backport/galactic)