ros2/rmw_cyclonedds #453: colcon build fail

fengtuo58, 2023-05-12:
## Bug report
/ros2_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_get_network_flow_endpoints.cpp:16:10: fatal error: rmw/get_network_flow_endpoints.h: No such file or directory
   16 | #include "rmw/get_network_flow_endpoints.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rmw_cyclonedds_cpp.dir/build.make:63: CMakeFiles/rmw_cyclonedds_cpp.dir/src/rmw_get_network_flow_endpoints.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/rmw_cyclonedds_cpp.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rmw_cyclonedds_cpp [0.14s, exited with code 2]

**Required Info:**
ros2  release version: foxy 
- Operating System:
- Linux byd-VirtualBox 5.15.0-71-generic #78~20.04.1-Ubuntu SMP Wed Apr 19 11:26:48 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

Splinter1984, 2023-05-12:
@fengtuo58. I would venture to guess what you are trying to build `rmw_cyclonedds_cpp` from a branch other than `foxy`. If I'm wrong, could you tell me which version of `ros2/rmw` and `ros2/rmw_cyclonedds_cpp` you are trying to build.

clalancette, 2023-05-12:
> @fengtuo58. I would venture to guess what you are trying to build `rmw_cyclonedds_cpp` from a branch other than `foxy`.

Yes, that is likely the case.

Please open questions like this on https://answers.ros.org, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

closed 2023-05-12 by clalancette