ros2/rmw_cyclonedds #459: multiple network interfaces can not work with multiple PC

Cmccm123, 2023-06-02:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 20.04
- Installation type:
  binaries
- Version or commit hash:
  Galactic
- DDS implementation:
  CycloneDDS  0.8.0

#### Steps to reproduce issue
I  have a network setup on robot:

- wlan0 has IP address of 192.168.2.0 - netmask 255.255.255.0
- eth0 has IP address of 172.30.1.0 - netmask 255.255.255.0

localPC:
- enp8s0 has IP address of 192.168.2.0 - netmask 255.255.255.0 (same network of robot's wlan0)

Robot Ip address: 172.30.1.24&192.168.2.48
My local pc ip address:192.168.2.61

my CycloneDDS setup: 
```<?xml version="1.0" encoding="utf-8"?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
  <Domain id="any">
    <General>
        <NetworkInterfaceAddress>eth0,wlan0</NetworkInterfaceAddress>`
        <AllowMulticast>false</AllowMulticast>
    </General>
    <Discovery>
      <ParticipantIndex>auto</ParticipantIndex>
      <Peers>
            <Peer Address="127.0.0.1" />
            <Peer Address="192.168.2.61" />
      </Peers>
      <MaxAutoParticipantIndex>120</MaxAutoParticipantIndex>
    </Discovery>
  </Domain>
</CycloneDDS>
```

Everything was working fine, but today the same settings suddenly not work :(

The situation is the same as #455. I used wireshark  in  local PC, found that the SPDP packets being sent over wlan0 are coming from 172.30.1.24

here is the CycloneDDS log. 

[log.zip](https://github.com/ros2/rmw_cyclonedds/files/11637871/log.zip)

#### Expected behavior

Only the local pc and the robot itself can view the ros topic of the robot on wlan0.
eth0 has no limit

#### Actual behavior
local pc can not  access ros topic of the robot.

#### Other Info

Please feel free to contact me if you need any further information.  Thank you for the special care you have given to the matter. :)

Best regards

Cmccm123, 2023-06-02:
update: The CycloneDDS log is provided by robot

clalancette, 2023-06-15:
Galactic is now End of Life.  Could you please try your setup with Humble or Iron and see if you have the same issue?  Thank you.

videh25, 2024-09-02:
Facing same error on Ubuntu 20.04 and ROS Foxy.

clalancette, 2024-09-03:
These are very old versions of ROS 2, that are no longer supported.  I'm going to close this out for now, but if you are still experiencing the issue on a supported version of ROS 2 (Humble, Iron, or Jazzy), please feel free to open another issue.

closed 2024-09-03 by clalancette

pietrobalatti, 2026-01-20:
Hi @clalancette, this has not been solved with humble, Im suggesting to reopen the issue. See also this [issue](https://github.com/eclipse-cyclonedds/cyclonedds/issues/1915). Having multiple NICs does not seem to be working,