ros2/rmw_cyclonedds #470: Add Iceoryx introspection tool to ROS Iceoryx build

ksuszka, 2023-09-25:
## Feature request

Please add Iceoryx introspection tools to ROS.

#### Feature description

Iceoryx comes with very useful introspection tool `iox-introspection-client` which shows current utilization of the shared memory transport.

#### Implementation considerations

Unfortunately, Iceoryx version in `ros-humble-iceoryx-*` packages is not compatible with `iox-introspection-client` installed by the `iceoryx` package.
Please add building `iox-introspection-client` to ROS build. AFAIK it can be added be enabling INTROSPECTION=ON cmake option during Iceoryx build.

eboasson, 2023-10-11:
A good suggestion, but I don't know who is responsible for maintaining the Iceoryx build in ROS, Cyclone just depends on it.

@clalancette do you know whether anyone is maintaining the build/packaging for Iceoryx in ROS?

clalancette, 2023-10-11:
> @clalancette do you know whether anyone is maintaining the build/packaging for Iceoryx in ROS?

The maintainer is listed as "iceoryx-oss-support@apex.ai"

eboasson, 2023-10-11:
Stupid me, could've thought of looking for that myself. Thanks @clalancette, I'll email them.

mossmaurice, 2023-11-17:
@ksuszka Thanks for creating the issue.

I have created the respective issue in the iceoryx repository (https://github.com/eclipse-iceoryx/iceoryx/issues/2099) and will add the introspection client.

cc @eboasson

mossmaurice, 2023-12-05:
@ksuszka 

The following releases for ROS 2 `humble`, `iron`, `rolling` and for iceoryx itself were created, which include the iceoryx introspection:

- https://github.com/ros/rosdistro/pull/39143
- https://github.com/ros/rosdistro/pull/39151
- https://github.com/ros/rosdistro/pull/39154
- https://github.com/eclipse-iceoryx/iceoryx/releases/tag/v2.0.5

Could you give it a try? Then we can close this issue.

cc @eboasson