ros2/rmw_cyclonedds #478: Error getting RMW implementation identifier

deepak-anzo, 2023-11-22:
## Bug report

 [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_cyclonedds_cpp'), with error message 'failed to load shared library 'librmw_cyclonedds_cpp.so' due to dlopen error: libddsc.so.0: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99, at ./src/functions.cpp:65', exiting with 1., at ./src/rcl/rmw_implementation_identifier_check.c:139 

**Required Info:**

- Operating System:
  -  OS and version  LUbuntu 22.04  ROS2 HUMBLE
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - <!  rclcpp -->

clalancette, 2023-11-22:
Can you give some information on what steps you followed to install Humble?

0116lhc, 2024-02-26:
I have the same situation,  you can try `sudo apt-get install -y ros-humble-rmw-cyclonedds-cpp `

fujitatomoya, 2024-02-26:
@deepak-anzo can you go through https://docs.ros.org/en/humble/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html? i think you are just missing the cyclone dds package.

deepak-anzo, 2024-02-27:
I found the solution for this. 
Just put `/opt/ros/humble/lib/x86_64-linux-gnu ` in your ld config file.

lqiang2003cn, 2025-05-14:
> I have the same situation, you can try `sudo apt-get install -y ros-humble-rmw-cyclonedds-cpp `

thank you very much, your answer saved my day. 
I might have accidentally deleted rmw while i was cleaning my system. Reinstall ros-humble-rmw-cyclonedds-cpp solved the problem.

jdgre1, 2025-08-07:
Yes this helped me too with my wsl with humble vscode setup!