ros2/rmw_cyclonedds #483: ROS2 communication stopped between nodes on same machine when network is down

robert-preissl, 2024-02-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.03
- Installation type:
  - binaries (running the latest ros2/iron docker image)
- Version or commit hash:
  - ros2 Iron
- DDS implementation:
  - ros-iron-cyclonedds 0.10.4-1jammy
  - ros-iron-rmw-cyclonedds-cpp 1.6.0-2jammy
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

The setup is as follow: using one computer, connected to a wifi network, two terminals both running a ros2 iron docker container
(the wireless interface as well as the loopback lo interface are configured as Multicast)

* run in one terminal the ros2 listener node (environment variables are the RMW_IMPLEMENTATION= rmw_cyclonedds_cpp):
`ros2 run demo_nodes_cpp listener`

* in a second terminal run the ros2 talker node (same environment variables as above):
`ros2 run demo_nodes_cpp talker`

* the terminal with the listener prints Hello World messages with increasing ids. (1,2,3, etc.)
* Now, we disconnect the wifi on the computer
* the terminal with the listener does not print new Hello world messages
* the terminal of the talker prints error messages like `tev: ddsi_upd_conn_write to dup/192.168.128.127:21163 failed with retcode -3`

#### Expected behavior

* since all the communication is local (i.e., on the same machine), an interruption in the wifi network should not interrupt local ROS communication

#### Actual behavior

* communication from the talker to the listener (all on the same host -- but different docker containers) is impacted.

wjwwood, 2024-03-07:
@eboasson may be able to comment here

robert-preissl, 2024-03-18:
Thank you @eboasson

Amit10311, 2025-02-03:
@eboasson @wjwwood @robert-preissl Hey ! I have same problem do you know the solution ?