ros2/rmw_cyclonedds #487: Ubuntu Host (Humble) and Docker (Iron) communication issue - [rmw_cyclonedds_cpp]: Failed to parse type hash for topic + xmlrpc.client.ResponseError: ResponseError("unknown tag 'rclpy.type_hash.TypeHash'")

carlos-argueta, 2024-04-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.4 LTS
- Installation type:
  - Binaries
- Version or commit hash:
  - Humble
- DDS implementation:
  - rmw_cyclonedds_cpp
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
At my company we are testing new features of ROS Iron, and the interoperability of new features in Iron with production features under Humble. To achieve this we want to run the Iron nodes in a Docker container to avoid messing up our working Humble setup. So, we have a host with Ubuntu 22.04 and Humble, and a Docker container with Iron. Both are using CycloneDDS by setting this variable:

_export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp_

And using the workaround described here #458 to fix the related issue.

Then I launch the container with:

_docker run -it --rm --network host -e CYCLONEDDS_URI=file:///usr/local/share/middleware_profiles/cyclonedds_profile.xml --name asc-iron asc-iron_

Followed by launching other nodes (simulation, navigation, etc) in my host.

#### Expected behavior
The expected behavior is to have a normal communication between the different nodes (host and container nodes). I should be able to echo topics published in the host from the container and vice-versa.

#### Actual behavior

As soon as the nodes on the host launch, the Docker container reports several messages (I believe one per topic) like:

[WARN] [1712819711.336704070] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'ros_discovery_info' with type 'rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_' from USER_DATA '(null)'.
[WARN] [1712819711.340353583] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'ros_discovery_info' with type 'rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_' from USER_DATA '(null)'.
[WARN] [1712819711.340785779] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/rosout' with type 'rcl_interfaces::msg::dds_::Log_' from USER_DATA '(null)'.
[WARN] [1712819711.341017792] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/parameter_events' with type 'rcl_interfaces::msg::dds_::ParameterEvent_' from USER_DATA '(null)'.
[WARN] [1712819711.341097649] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'ros_discovery_info' with type 'rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_' from USER_DATA '(null)'.
[WARN] [1712819711.341221168] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/parameter_events' with type 'rcl_interfaces::msg::dds_::ParameterEvent_' from USER_DATA '(null)'.
[WARN] [1712819711.347964180] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/parameter_events' with type 'rcl_interfaces::msg::dds_::ParameterEvent_' from USER_DATA '(null)'.
[WARN] [1712819711.349610842] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/parameter_events' with type 'rcl_interfaces::msg::dds_::ParameterEvent_' from USER_DATA '(null)'.
[WARN] [1712819711.355084576] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/sim/imu' with type 'sensor_msgs::msg::dds_::Imu_' from USER_DATA '(null)'.
[WARN] [1712819711.356171394] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/clock' with type 'rosgraph_msgs::msg::dds_::Clock_' from USER_DATA '(null)'.
....

I log into the container with _docker exec -it asc-iron bash_ then I source the relevant workspaces and list the topics with _ros2 topic list_ and the list of topics show up.
When I try to see the data from a topic with _ros2 topic echo <topic_name>_ no data comes out.

I then go to the host and try to see the data for the same topic (with same _ros2 topic echo_ ), to confirm that there is indeed data, and I get this error:

Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.9', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/verb/echo.py", line 220, in main
    qos_profile = self.choose_qos(node, args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/verb/echo.py", line 146, in choose_qos
    pubs_info = node.get_publishers_info_by_topic(args.topic_name)
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1122, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1464, in __request
    response = self.__transport.request(
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1166, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1182, in single_request
    return self.parse_response(resp)
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1348, in parse_response
    p.feed(data)
  File "/usr/lib/python3.10/xmlrpc/client.py", line 451, in feed
    self._parser.Parse(data, False)
  File "../Modules/pyexpat.c", line 416, in StartElement
  File "/usr/lib/python3.10/xmlrpc/client.py", line 689, in start
    raise ResponseError("unknown tag %r" % tag)
xmlrpc.client.ResponseError: ResponseError("unknown tag 'rclpy.type_hash.TypeHash'")

Strangely, if I stop the container and I immediately run the echo command again, I get this:

WARNING: topic [/topic_name] does not appear to be published yet
Could not determine the type for the passed topic

and if I run the command a moment later, the data comes normally.

carlos-argueta, 2024-04-12:
> This might be relevant: https://answers.ros.org/question/341372/can-nodes-from-different-ros-2-distributions-communicate-compatibly/

Absolutely relevant thanks. Now the issue might have to do with how CycloneDDS handles the exchange. For instance, with Fast-DDS I can actually get the data from both ends, but then when some particular topics are echoed the memory consumption goes up until the computer (32 GB RAM) is out of resources and all nodes are killed by the system.

It seems to me that the CycloneDDS issue could be more manageable. Following the answers on your suggested post, I checked the changes from the release page of Iron. A potentially relevant paragraph says:

_The full TypeDescription data structure, as well as the raw source text (such as .msg file) that were used to generate it are now baked in by default to the message libraries, so they can be used by typesupport or end users. While we expect this data to provide value to most users, some users trying to minimize bytes in their install space can disable the feature when building ROS 2 Core by defining the CMake variable ROSIDL_GENERATOR_C_DISABLE_TYPE_DESCRIPTION_CODEGEN._

The issue I see on the host side using CycloneDDS is that the host does not know what type_hash.TypeHash is (_xmlrpc.client.ResponseError: ResponseError("unknown tag 'rclpy.type_hash.TypeHash'")_). Could building ROS 2 with variable ROSIDL_GENERATOR_C_DISABLE_TYPE_DESCRIPTION_CODEGEN._ turned on fix the issue?

I will have to try.

MichaelOrlov, 2024-04-25:
@carlos-argueta Please close this issue since we wouldn't fix it because there is no guarantee to have cross-distro compatibility. 
For the reasons that sometimes we have to introduce compatibility-breaking changes to support new features in the new releases.

carlos-argueta, 2024-04-29:
> @carlos-argueta Please close this issue since we wouldn't fix it because there is no guarantee to have cross-distro compatibility. For the reasons that sometimes we have to introduce compatibility-breaking changes to support new features in the new releases.

Hi Michael, sorry I was meaning to this a lot earlier, and I do understand the situation. Thanks for your message.

closed 2024-04-29 by carlos-argueta