ros2/rmw_cyclonedds #503: ROS service not receiving native DDS requests correctly

tmayoff, 2024-07-09:
## Bug report

**Required Info:**

- Operating System:
  - debian bookworm
  - ubuntu 22.04 (docker)
- Installation type:
  - binaries
- Version or commit hash:
  - ros2 humble
- DDS implementation:
  - cyclonedds
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
1. Setup node and service 

```c++
Transformer::Transformer() : rclcpp::Node("transformer"), tf_listener(nullptr) {
  this->get_logger().set_level(rclcpp::Logger::Level::Debug);

  tf_buffer = std::make_unique<tf2_ros::Buffer>(this->get_clock());
  tf_listener = std::make_shared<tf2_ros::TransformListener>(*tf_buffer);

  raycast_pose = this->create_publisher<geometry_msgs::msg::PoseStamped>("/looking_at", 10);

  transform_service = this->create_service<avatar_msgs::srv::TransformPoint>(
      "transformer", [this](const avatar_msgs::srv::TransformPoint::Request::SharedPtr request,
                            avatar_msgs::srv::TransformPoint::Response::SharedPtr response) { service_callback(request, response); });
}

int main(int argc, char** argv) {
  rclcpp::init(argc, argv);

  auto node = std::make_shared<Transformer>();
  rclcpp::spin(node);
  rclcpp::shutdown();

  return 0;
}
```

```.srv
string source_frame
string target_frame
geometry_msgs/Transform input
---
geometry_msgs/Transform output
bool valid
```

```idl
#ifndef TRANSFORMER_REQUEST_IDL
#define TRANSFORMER_REQUEST_IDL

#include "geometry_msgs/msg/Transform.idl"

module avatar_msgs {
  module srv {
    module dds_ {
      @final @topic struct TransformPoint_Request_ {
        string source_frame;
        string target_frame;
        geometry_msgs::msg::dds_::Transform_ input;
      };
    };
  };
};
[... 101 lines cut ...]
│ │ │ LatencyBudget(budget='zero')                                                                                                         │ │ │
│ │ │ Lifespan(lifespan='infinity')                                                                                                        │ │ │
│ │ │ Liveliness.Automatic(lease_duration='infinity')                                                                                      │ │ │
│ │ │ Ownership.Shared                                                                                                                     │ │ │
│ │ │ OwnershipStrength(strength=0)                                                                                                        │ │ │
│ │ │ PresentationAccessScope.Instance(coherent_access=False, ordered_access=False)                                                        │ │ │
│ │ │ Reliability.Reliable(max_blocking_time='100 milliseconds')                                                                           │ │ │
│ │ │ ResourceLimits(max_samples=-1, max_instances=-1, max_samples_per_instance=-1)                                                        │ │ │
│ │ │ TransportPriority(priority=0)                                                                                                        │ │ │
│ │ │ WriterDataLifecycle(autodispose=True)                                                                                                │ │ │
│ │ ╰──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────╯ │ │
│ │ ┌────────────────┬─────────────────────────────────────────────────┐                                                                     │ │
│ │ │ Typename       │ avatar_msgs::srv::dds_::TransformPoint_Request_ │                                                                     │ │
│ │ │ XTypes Type ID │           COMPLETE 7AB8E5650C79612B62E1F36D1635 │                                                                     │ │
│ │ └────────────────┴─────────────────────────────────────────────────┘                                                                     │ │
│ │               Publications                                                                                                               │ │
│ │ ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓                                                                                                 │ │
│ │ ┃ GUID                                 ┃                                                                                                 │ │
│ │ ┡━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩                                                                                                 │ │
│ │ │ 01101a2e-0d3e-ba49-5cea-892e00000203 │                                                                                                 │ │
│ │ └──────────────────────────────────────┘                                                                                                 │ │
│ ╰──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────╯ │
╰──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────╯

```

tmayoff, 2024-07-10:
So I implemented the service in ROS using pub/subs instead and it works with the same client code. I guess there's something I'm missing when setting up the clients (or probably less likely there's a bug here?) any help would be great

eboasson, 2024-07-23:
@tmayoff could it be that your native DDS types don't account for the request/response headers that the Cyclone RMW layer injects?

The header type is
```
struct RequestHeader {
  uint64_t guid;
  int64_t seq;
};
```
The `guid` and `seq` just need to be unique.