ros2/rmw_cyclonedds #511: Install a reasonable default change to `rmem_max` when installing the CycloneDDS RMW

EzraBrooks, 2024-09-06:
## Feature request

#### Feature description

It's a common hurdle that new users of the CycloneDDS RMW find their topic subscribers grinding to a halt once they plug in, for example, a couple Intel RealSense RGBD cameras - or even a simulated equivalent in something like Gazebo. In my view, streaming depth data from a couple cameras is a pretty basic requirement for a robotic system, and this RMW should be robust to that out-of-the-box.

#### Implementation considerations

The [CycloneDDS tuning](https://docs.ros.org/en/jazzy/How-To-Guides/DDS-tuning.html#cyclone-dds-tuning) section of the ROS documentation instructs users to create an `/etc/sysctl.d/10-cyclone-max.conf` file. It would be simple to install a default version of that file as part of the Debian package installer for the RMW. While I'm usually not a proponent of doing things like modifying a user's kernel parameters on their behalf, it seems pretty clear that this is required for this RMW to function optimally.

The hardest part of this task will be deciding on what a "reasonable value" is, but I have yet to experience any issues with the recommended value in the documentation, so we could just use that.

clalancette, 2024-09-06:
> While I'm usually not a proponent of doing things like modifying a user's kernel parameters on their behalf, it seems pretty clear that this is required for this RMW to function optimally.

I'm really against this; I really don't think ROS 2 should be changing the kernel parameters.

That said, we could escalate this from just a documentation thing, to something that the code looks at startup time and prints an INFO/WARN about.  That would at least make the user more likely to go look at the documentation.

EzraBrooks, 2024-09-06:
That sounds like a reasonable thing to do. The only reason I suggested including the config file in the package is because this has bitten so many people that in my project we _did_ put the config file into our installer because otherwise every. single. user. ran into the issue and panicked instead of reading ROS documentation on how to fix the issue. I figure there's probably others out there doing exactly the same thing as I am who would benefit from it being upstreamed.

But adding a warning sounds nice if the project is staunchly against installing a file into `/etc/sysctl.d/`.

For what it's worth, I don't think it's _that_ odd for something as low-level as a network protocol to modify kernel parameters (I know technically this is like two layers of abstraction up from being a protocol, but :shrug:).

EzraBrooks, 2024-09-06:
Given that the documentation suggests dropping that config into a file, another option would be to put that file in the place the documentation says to put it, but have it contain the parameters commented out so the user action is to "affirm they want this change" by uncommenting it.

clalancette, 2024-09-06:
The other major problem we have is that we always install things into `/opt/ros/<rosdistro>`.  Putting something in `/etc` breaks that model (and I don't even know if our tooling supports it).  For instance, if you happen to install e.g. both Jazzy and Rolling on the same machine, which one should own the file in `/etc/sysctl.d`?

EzraBrooks, 2024-09-06:
Ah, good point. It's probably best not to violate that convention in something as core as an RMW. The only potential solution I could see to that would be using the distro version in the name of the configuration file, but that feels sketchy.

EzraBrooks, 2024-09-06:
Opened #512 regarding logging a warning

closed 2024-09-06 by EzraBrooks