ros2/rmw_cyclonedds #514: Is there a way to obtain the address of shared memory at init phase ?

lix19937, 2024-09-24:
The loan API allows for true zero-copy transfer by directly consctructing the message data in Shared Memory. To do so, we first have to loan this Shared Memory chunk explicitly.

```cpp  
// init phase 
std::set<void *> ptr_set;

// in callback phase at runtime  
auto loaned_msg = pub_->borrow_loaned_message();
loaned_msg.get().data = count_++;
pub_->publish(std::move(loaned_msg));
ptr_set.insert(&(loaned_msg.get().data));
```

I  found  `ptr_set.size() = N `

fujitatomoya, 2024-10-04:
@lix19937 can you close this issue? this is more like a question, please consider to post this kind of question to https://robotics.stackexchange.com/ where is the centralized QA site. it is likely that you can get the feedback against the questions from community.

lix19937, 2024-10-04:
OK, thanks  @fujitatomoya

closed 2024-10-04 by lix19937