ros2/rmw_cyclonedds #535: 🧑‍🌾

Crola1702, 2025-03-28:
### Operating System:

RHEL 9

### ROS version or commit hash:

Rolling

### RMW implementation (if applicable):

rmw_cyclonedds_cpp

### 'ros2 doctor --report' output

```Formatted

```

### Steps to reproduce issue

1. Run a build in RHEL CI
2. See downstream test regressions fail

### Expected behavior

Tests should pass

### Actual behavior

Test regressions fail

### Additional information

Reference builds:
* [nightly_linux-rhel_repeated#2234](https://ci.ros2.org/job/nightly_linux-rhel_repeated/2234)
* [nightly_linux-rhel_debug#2239](https://ci.ros2.org/job/nightly_linux-rhel_debug/2239)
* [nightly_linux-rhel_release#2222](https://ci.ros2.org/job/nightly_linux-rhel_release/2222)

<details>
<summary>
Test regressions:
</summary>

* [projectroot.test_tutorial_content_filtering_publisher_content_filtering_subscriber__rmw_cyclonedds_cpp](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/(root)/projectroot/test_tutorial_content_filtering_publisher_content_filtering_subscriber__rmw_cyclonedds_cpp/)
* [demo_nodes_cpp.TestExecutablesTutorial.test_processes_output](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/demo_nodes_cpp/TestExecutablesTutorial/test_processes_output/)
* [demo_nodes_cpp.TestExecutablesTutorialAfterShutdown.test_last_process_exit_code](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/demo_nodes_cpp/TestExecutablesTutorialAfterShutdown/test_last_process_exit_code/)
* [projectroot.test_manual_composition__rmw_cyclonedds_cpp](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/(root)/projectroot/test_manual_composition__rmw_cyclonedds_cpp/)
* [composition.TestComposition.test_manual_composition](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/composition/TestComposition/test_manual_composition/)
* [projectroot.test.rclcpp.test_add_callback_groups_to_executor__rmw_cyclonedds_cpp](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/projectroot.test/rclcpp/test_add_callback_groups_to_executor__rmw_cyclonedds_cpp/)
* [rclcpp.test_add_callback_groups_to_executor.gtest.missing_result](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/(root)/rclcpp/test_add_callback_groups_to_executor_gtest_missing_result/)
* [projectroot.test_play_next__rmw_cyclonedds_cpp](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/(root)/projectroot/test_play_next__rmw_cyclonedds_cpp/)
* [rosbag2_transport.test_play_next.gtest.missing_result](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/(root)/rosbag2_transport/test_play_next_gtest_missing_result/)
* [ros2param.ros2param.test.test_verb_dump.test_verb_dump](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/ros2param.ros2param.test/test_verb_dump/test_verb_dump/)
* [projectroot.test_play_timing__rmw_cyclonedds_cpp](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/(root)/projectroot/test_play_timing__rmw_cyclonedds_cpp/)
* [ros2service.ros2service.test.test_cli.test_cli](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/ros2service.ros2service.test/test_cli/test_cli/)
* [intra_process_demo.TestExecutablesDemo.test_executables_output](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/intra_process_demo/TestExecutablesDemo/test_executables_output/)
* [ros2doctor.ros2doctor.test.test_qos_compatibility.test_qos_compatibility](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/ros2doctor.ros2doctor.test/test_qos_compatibility/test_qos_compatibility/)
* [sros2.test.sros2.commands.security.verbs.test_generate_policy.test_generate_policy](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/sros2.test.sros2.commands.security.verbs/test_generate_policy/test_generate_policy/)
* [rosbag2_transport.test_play_timing.gtest.missing_result](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/(root)/rosbag2_transport/test_play_timing_gtest_missing_result/)
* [ros2topic.ros2topic.test.test_cli.test_cli](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/ros2topic.ros2topic.test/test_cli/test_cli/)
* [sros2.test.sros2.commands.security.verbs.test_generate_policy_no_nodes.test_generate_policy_no_nodes](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/sros2.test.sros2.commands.security.verbs/test_generate_policy_no_nodes/test_generate_policy_no_nodes/)
* [ros2node.ros2node.test.test_cli.test_cli](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/ros2node.ros2node.test/test_cli/test_cli/)
* [rclcpp.test_add_callback_groups_to_executor.gtest.missing_result](https://ci.ros2.org/job/nightly_linux-rhel_release/2222/testReport/junit/(root)/rclcpp/test_add_callback_groups_to_executor_gtest_missing_result/)

</details>

Tests are showing the following problem:
```
1743143011.718972 [1] dq.builtin: type [INVALID 0000000000000000000000000000]: ddsi_xt_type_init_impl with invalid type object
```

Crola1702, 2025-04-02:
Reproducing the problem:

* Should fail: [image]
* Should pass: [image]

cottsay, 2025-04-15:
I can reproduce this on Fedora 41 as well, though not consistently.

fujitatomoya, 2025-04-29:
@eboasson can you please share your thoughts what could possibly happen here when you have time? this can be the problem for  new release Kilted Kaiju test party to test `rmw_cyclonedds_cpp`.

fujitatomoya, 2025-04-29:
with my local test environment (ubuntu noble container userspace, Linux 5.15.0-134-generic, 11th Gen Intel(R) Core(TM) i9-11900K @ 3.50GHz), i cannot reproduce this issue with the following commands.

- terminal-1

```bash
root@tomoyafujita:/# stress-ng --cpu 0 --timeout 30m
```

- terminal-2

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# colcon test --event-handlers console_direct+ --packages-select rosbag2_transport --ctest-args -R test_record_regex__rmw_cyclonedds_cpp --repeat-until-fail 100
```

- terminal-3

```bash
colcon test --event-handlers console_direct+ --packages-select demo_nodes_cpp --ctest-args -R test_tutorial_talker_listener__rmw_cyclonedds_cpp --repeat-until-fail 100
```

Blast545, 2025-06-25:
🧑‍🌾 We discussed in the ROS PMC meeting today that Scott is adding isolation around rmw implementations, which might improve understanding if the problem happens at the rmw layer or elsewhere. So far the team has had trouble reproducing this problem.

Crola1702, 2025-07-11:
I'm seeing the same behavior in Windows repeated: [test_security.TestSecurePublisherSubscriberAfterShutdown.test_processes_finished_gracefully](https://ci.ros2.org/job/nightly_win_rep/3792/testReport/junit/test_security/TestSecurePublisherSubscriberAfterShutdown/test_processes_finished_gracefully/)

FYI: @cottsay

Crola1702, 2025-07-16:
`rosbag2_transport` also has this issue on Windows. See [nightly_win_rep#3796](https://ci.ros2.org/job/nightly_win_rep/3796/testReport/junit/(root)/projectroot/test_record_all_include_unpublished_topics__rmw_cyclonedds_cpp/):

```
[ RUN      ] RecordIntegrationTestFixture.record_all_include_unpublished_false_includes_later_published
stdin is not a terminal or console device. Keyboard handling disabled.[INFO] [1752570001.398466200] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1752570001.400241300] [rosbag2_recorder]: Listening for topics...
[INFO] [1752570001.400340700] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1752570001.401505500] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[INFO] [1752570001.403495100] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[INFO] [1752570001.404268800] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1752570001.404311800] [rosbag2_recorder]: Recording...
[INFO] [1752570001.507031600] [rosbag2_recorder]: Subscribed to topic '/string_topic'
[INFO] [1752570001.611556800] [rosbag2_recorder]: Subscribed to topic '/lc_talker/transition_event'
[INFO] [1752570001.813985500] [rosbag2_recorder]: Pausing recording.
[INFO] [1752570001.816948400] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1752570001.817170600] [rosbag2_recorder]: Recording stopped
1752570002.070406 [1] dq.builtin: type [INVALID 0000000000000000000000000000]: ddsi_xt_type_init_impl with invalid type object
-- run_test.py: return code 3221225477
-- run_test.py: generate result file 'C:/ci/ws/build/rosbag2_transport/test_results/rosbag2_transport/test_record_all_include_unpublished_topics.gtest.xml' with failed test
-- run_test.py: verify result file 'C:/ci/ws/build/rosbag2_transport/test_results/rosbag2_transport/test_record_all_include_unpublished_topics.gtest.xml'
```

Also in `test_communication` on linux repeated. See [nightly_linux_repeated#3907]([projectroot.test_publisher_subscriber__rclcpp__rmw_cyclonedds_cpp__rmw_fastrtps_cpp](https://ci.ros2.org/job/nightly_linux_repeated/3907/testReport/junit/(root)/projectroot/test_publisher_subscriber__rclcpp__rmw_cyclonedds_cpp__rmw_fastrtps_cpp/))

```
Starting test run test_communication.test_publisher_subscriber__rclcpp__rmw_cyclonedds_cpp__rmw_fastrtps_cpp_Release.launch_tests[Empty]
[INFO] [launch]: All log files can be found below /home/rosbuild/.ros/log/2025-07-15-21-51-28-184088-48dc26fb818d-604333
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [daemon-stop-22]: process started with pid [604582]
[INFO] [daemon-stop-22]: process has finished cleanly [pid 604582]
test_subscriber_terminates_in_a_finite_amount_of_time[Empty] (test_communication.TestPublisherSubscriber.test_subscriber_terminates_in_a_finite_amount_of_time[Empty])
Test that the subscriber terminates after a finite amount of time. ... [INFO] [test_publisher-23]: process started with pid [604585]
[INFO] [test_subscriber-24]: process started with pid [604586]
[test_publisher-23] 1752616293.930554 [85] dq.builtin: type [INVALID 0000000000000000000000000000]: ddsi_xt_type_init_impl with invalid type object
[ERROR] [test_publisher-23]: process has died [pid 604585, exit code -11, cmd '/home/jenkins-agent/workspace/nightly_linux_repeated/ws/build/test_communication/test_publisher_cpp Empty /test_time_21_51_33'].
FAIL
```

FYI: @cottsay

cottsay, 2025-07-16:
I'm still not 100% convinced that this issue is caused by crosstalk among tests. At the very least, the fact that we only observe the problem on `rmw_cyclonedds_cpp` indicates that there is likely isn't a systematic problem with our approach to testing, but rather a Cyclone-specific behavior that we have yet to understand.

None the less, I've been rolling out the test-level isolation in `rolling`, which is more robust than the existing package-level isolation. Some reference PRs:
- ros2/system_tests#571
- ros2/demos#724
- ros2/ros2cli#1062

Crola1702, 2025-09-02:
I don't see this happening anymore, so closing. Thanks to everyone that worked on this!

closed 2025-09-02 by Crola1702